Duke-BME-Design/Aiguille-Medicale #8: Determine Best Circuit

EddieYao8, 2019-01-15:
Test other possible circuits for metal detection

ChanceFleeting, 2019-01-22:
At end of last semester: Bias towards VLF based on "achieved" efficacy and ease of bipolar design, but still low discount on PI if achievable.  Cheap Metal detector will give a jump-start to this.

ChanceFleeting, 2019-02-02:
Moving this to back log because, while we are sitting on what seems to be **Bipolar VLF**, we need to look into modes of amplification within circuit design. Since we are moving over our hour count and I don't want to rush this, I will stall it.

ChanceFleeting, 2019-02-08:
Moving this to review since so long as we are looking into Bipolar configuration, VLF is effectively our only choice.