Duke-BME-Design/Aiguille-Medicale #15: Mock up meat + needle detection test

kaylawf, 2019-01-16:


AznStevy, 2019-01-16:
@mlp6 @sphaerobolus Just a question about logistics. If we make something like this, how will we store it? Is there a fridge at Chesterfield that we can use?

sphaerobolus, 2019-01-17:
I've asked my friends in the Lynch lab downstairs if they'd be willing to house your meat in their +4C fridge. They're hesitant, but willing. Looks like you'll be able to use the drawer in the attached(?) pic. Of course it goes without saying that their research materials come first and anything you do you mess them up will result in eviction. 

[image]