Duke-BME-Design/Aiguille-Medicale #24: Make Receiver Pad coil design

EddieYao8, 2019-01-24:


ChanceFleeting, 2019-01-25:
What is the minimum trace size/spacing available via photolithography?

ChanceFleeting, 2019-01-26:
I think that we could work with 20 mil with 10 mil gaps. We are trying to print planar inductors as a receiver for a bipolar metal detector, so the more turns we can fit, the higher precision/power we can get. The finer, the better. I'll look into the injectorall!

Thank you!

ChanceFleeting, 2019-01-29:
https://github.com/Duke-BME-Design/Aiguille-Medicale/tree/master/Electrical/Coil%20Design/Receiver

ChanceFleeting, 2019-02-02:
@sphaerobolus, I think I have the the traces at 20 mil with 10 mil gaps. Currently the two designs I have are a single 6-in-diameter spiral with roughly 66 turns and an 3-by-3 array of 2-in-diameter coils each with about 25 turns are there any other specific considerations I should make in designing a photomask for the flexible kapton stuff.

Thank you, again!

https://github.com/Duke-BME-Design/Aiguille-Medicale/tree/master/Electrical/Coil%20Design/Receiver

sphaerobolus, 2019-02-03:
Hm. I should have thought longer and written a bit more in my comments, above. I think we have hope of doing this, but I'm not more than 50% confident it will work. With curved traces running at our size and separation limit for so long, our chances of problems go way up. I'm willing to try it. If you'd like to, I'll get the materials on the list to order. Doing a quick price check, this looks like roughly a $300 job to have a board shop fabricate on flexible PCB.

ChanceFleeting, 2019-02-05:
Oh that's pretty steep for a prototype. I'm going to look into seeing it would be a viable move to spiral coil onto a flat sticky surface by hand using the 42 gauge that we have... Are there any recommendation on sticky, flexible, magnetically inert, nonconductive material for me to spiral these coils onto?

ChanceFleeting, 2019-02-08:
I'm making a few templates to make spiral inductors to try and coil this weekend/Friday. I think that I can get it to an acceptable clearance for 18 AWG. What gauge magnetic wire do you have?

Also Kapton sound awesome as a backing.

ChanceFleeting, 2019-02-12:
@sphaerobolus Curiosity, what would you put the price point/viability of photomasking one of the at-resolution coils onto a 6"x6" or 2"x2" rigid board? for the demo, if we edit the experiment a little, we could potentially get by with a rigid reciever.

Thank you again for all your help!

sphaerobolus, 2019-02-12:
It's certainly worth a try
what's your timeline? I will etch a bunch of boards next week, and it would be convenient to lump this job with those.

ChanceFleeting, 2019-02-12:
If these two designs look good as masks, then I think we are ready whenever is best for you.

ChanceFleeting, 2019-02-12:
https://github.com/Duke-BME-Design/Aiguille-Medicale/blob/master/Electrical/Coil%20Design/Receiver/Photolith_Multi.png

https://github.com/Duke-BME-Design/Aiguille-Medicale/blob/master/Electrical/Coil%20Design/Receiver/Photolith_Sing.png

sphaerobolus, 2019-02-12:
PNG doesn't give me exact scale. Can you give me an image, e.g. PDF, at 1:1 scale?

ChanceFleeting, 2019-02-12:
I'm not sure how to check the scale factor, but I moved them over to a pdf.
https://github.com/Duke-BME-Design/Aiguille-Medicale/blob/master/Electrical/Coil%20Design/Receiver/Photolith.pdf

I also have verification that the printed trace width and gap ought to be 20 and 10 mil respectively.
https://github.com/Duke-BME-Design/Aiguille-Medicale/tree/master/Electrical/Coil%20Design/Receiver/Validation

sphaerobolus, 2019-02-12:
Check the distance from pad to pad in your PDF. Scale as needed to make this distance the designed distance.

ChanceFleeting, 2019-02-19:
@sphaerobolus, do you mean that I should print it out and measure or is there a way to determine distance within a PDF?

sphaerobolus, 2019-02-19:
There might be a way to do it in silico, but the only way I know is on paper.