Duke-BME-Design/Aiguille-Medicale #50: Draw circuit for housing.

AznStevy, 2019-02-01:


AznStevy, 2019-02-08:
To get wall power, we need to have a ACDC converter + a full-wave rectifier. We'll probably need a step down transformer as well if we don't want to blow things up. We need 5 V going to the arduino nano, so we'll probably bring it down to 5 V and stick it into the raw voltage port. There needs to be high current going through the coil, so we'll have a fairly small resistance (maybe a power resistor to dissipate heat? + the housing will be cooled by the fan); this will also need some kind of amplifier and will eventually going into an analog port in the arduino. The fan will be driven by one of the ports on the arduino. For the handle, we need to have a few resistors going to ground and different ports on the arduino. The haptic motor will go directly into an analog output on the arduino. The reciever pad will also need some amplifier and will go into an analog port. Drawing coming soonish.

AznStevy, 2019-02-08:
We kinda want higher voltage, but we'll decide that in a bit i guess.
[image]

The current list of items seems to be:

- ACDC converter
- Diodes
- 5V regulator
- Arduino Nano
- Power resistor
- Opamps (RMS amplifier for receiver)
- Resistors (of whatever values)
- Relay

Add to this as you see fit.

ChanceFleeting, 2019-02-08:
Hey Stephen, For the detector circuitry, were we not doing to use the synchronous phase/Amplitude detection circuitry we specified last semester?

AznStevy, 2019-02-08:
Oh, i forgot about that. Though, I'm not sure how to immediately implement that lol

ChanceFleeting, 2019-02-08:
Kayla and I are on it

ChanceFleeting, 2019-02-09:
Also missing in your sketch:
fuse (lets put that up before we blow our circuit)
regulation (zener diodes and Voltage regulators)
energy storage (ie capacitors; otherwise this thing will just output a FWR wave and not anything remotely DC)

ChanceFleeting, 2019-02-09:
@AznStevy Also, why not just buy a ACDC converter with the power rating we need as power supply? I thought that's what we were going to do

AznStevy, 2019-02-09:
We can do that. I wasn't sure what was at chesterfield and what wasn't.

ChanceFleeting, 2019-02-09:
Tomorrow, lets try and get a gauge for the current rating we need to get through the Tx coil and work with that. Sensing is priority number 1, so we can use lab power until we tie that down. If you could sketch up/ build the Relay Driver circuit for VLF current switching, that would probably be the best at this time. Kayla is working on a schematic for the phase detector circuit, and I have already provided a a preliminary schematic for the Rx amplifier and an Amplitude detector circuit.