Duke-BME-Design/Aiguille-Medicale #112: Verify Mask pdf Dimensions

ChanceFleeting, 2019-02-20:
Measuring the print out of the mask to reverify that it is 1-to-1. Note trace widths were shown to be compliant with previous iteration.

ChanceFleeting, 2019-02-24:
@sphaerobolus , The PDF is approximately 1-to-1 with the desired dimensions on my side. For some reason, when I got rid of the boundaries, it magnified a little, but that shouldn't be too big of an issue.

https://github.com/Duke-BME-Design/Aiguille-Medicale/blob/master/Electrical/Coil%20Design/Receiver/Photolith_Multi.pdf

closed 2019-02-24 by ChanceFleeting

sphaerobolus, 2019-02-25:
Alas, still not ready for fabrication for two reasons:
1) Resolution was lost in whatever steps you took to scale it. This is not a spiral, but literally one large mass of copper, perforated occasionally with gray--not white--pixels.

2) We'll need at least 10 mils gap between copper traces, and again the gap must be white, not gray.